jourdant/esphome-lgap #1: Interesting project!

bzumik1, 2024-02-17:
Hi I plan to buy LG UM36F + UUC1 Comfort. As soon as I will have them I can help with the testing!

bzumik1, 2024-02-21:
I hope so, it would be really nice to have a way how to control the AC with ESPHome. Based on what I have read, the physical layer is passed on RS485 right so something like [M5Stack Official ATOMIC RS485 Base](https://a.aliexpress.com/_EveDCtF) will be perfect fit for this right?

jourdant, 2024-02-21:
That's right, if there is a Central Control header, that generally indicates there is either an RS485 interface on board that speaks LGAP, or there are pins you can connect an official PI-485 board to enable that interface also (generally older devices).

I'm a big fan of the M5Stack devices but my favourite device so far [LILYGO T-RSC3](https://www.lilygo.cc/products/t-rsc3). I have used this with esphome without issues.

I've got enough of the LGAP protocol decoded now to write a native esphome library. I've started work on that now.

bzumik1, 2024-02-21:
The LILYGO T-RSC3 looks really nice. I wasn't aware of this. Thanks for the tip. One problem I have with it is that it does not come with case 😮‍💨 

You are really that far? That is great. I guess simple "turn on/off" is not that far away!

jourdant, 2024-02-22:
If you're open to 3D printing a case, Lilygo actually have uploaded a printable case here: https://github.com/Xinyuan-LilyGO/T-RSC3/tree/main/Shell_file. The other thing I really liked about this board was that it can take a variable DC input and my LG ODU has a 12v terminal that isn't being used. That means I can wire up the device entirely directly to the ODU without needing any other external components for power supply.

Otherwise the M5stack boards are really nice. I haven't checked the RS485 interface on that board, but it's highly likely that it would work just fine.

jourdant, 2024-03-04:
Quick update - my [feature/esphome2](https://github.com/jourdant/esphome-lgap/tree/feature/esphome2) branch is working nicely now showing in HA, should hopefully have something ready for wider testing very soon.

grydo, 2024-03-22:
> Quick update - my [feature/esphome2](https://github.com/jourdant/esphome-lgap/tree/feature/esphome2) branch is working nicely now showing in HA, should hopefully have something ready for wider testing very soon.

Hi I would like to use lilygo t-rsc3 for 2 projects one with rs232 and one with rs485, I was trying rs485 with esphome apparently it doesn't work, could you give me some help?

this is what I need to insert into rs485

captive_portal:

button:
    - platform: restart
      id: mppt4880
      name: "${friendly_name} Restart"

uart:
  tx_pin: GPIO3
  rx_pin: GPIO1
  id: mod_bus
  baud_rate: 9600
  stop_bits: 1

modbus:
  #flow_control_pin: 23
  uart_id: mod_bus
  id: mod_bus_mppt

modbus_controller:
  - id: mppt
    ## the Modbus device addr
    address: 0x1
    modbus_id: mod_bus_mppt
    command_throttle: 200ms
    setup_priority: -10
    update_interval: 5s

sensor:
#   - platform: modbus_controller
#     modbus_controller_id: mppt

#     name: Rated Current
#     address: 10009
#     register_type: holding
#     value_type: U_WORD
#     register_count: 1
#     filters:
#       - multiply: 0.1
  - platform: modbus_controller
    modbus_controller_id: mppt

    name: Charger Workstate
    address: 15201
    register_type: holding
    value_type: U_WORD
    register_count: 1
#   - platform: modbus_controller
#     modbus_controller_id: mppt

#     name: MPPT state
#     address: 15202
#     register_type: holding
#     value_type: U_WORD
#     register_count: 1

#   - platform: modbus_controller
#     modbus_controller_id: mppt

#     name: Charging state
#     address: 15203
#     register_type: holding
#     value_type: U_WORD
#     register_count: 1

  - platform: modbus_controller
    modbus_controller_id: mppt
    id: pv_input_voltage
    name: pv_input_voltage
    address: 15205
    unit_of_measurement: "V" ## for any other unit the value is returned in minutes
    register_type: holding
    value_type: U_WORD
    register_count: 1

    accuracy_decimals: 1
    icon: mdi:sine-wave
    filters:
      - multiply: 0.1

  - platform: modbus_controller
    modbus_controller_id: mppt
    id: batt_input_voltage
    name: batt_input_voltage
    address: 15206
    unit_of_measurement: "V" ## for any other unit the value is returned in minutes
    register_type: holding
    value_type: U_WORD
    register_count: 1

    accuracy_decimals: 1
    icon: mdi:sine-wave
    filters:
      - multiply: 0.1
  - platform: modbus_controller
    modbus_controller_id: mppt
    id: batt_input_current
    name: batt_input_current
    address: 15207
    unit_of_measurement: "A"
    register_type: holding
    value_type: U_WORD
    register_count: 1
    filters:
      - multiply: 0.1
    accuracy_decimals: 1

  - platform: modbus_controller
    modbus_controller_id: mppt
    id: pv_input_power
    name: pv_input_power
    address: 15208
    unit_of_measurement: "W" ## for any other unit the value is returned in minutes
    register_type: holding
    value_type: U_WORD
    register_count: 1

    accuracy_decimals: 1
    icon: mdi:sine-wave

  - platform: modbus_controller
    modbus_controller_id: mppt

    name: MPPT temperature
    address: 15209
    unit_of_measurement: "°C" ## for any other unit the value is returned in minutes
    register_type: holding
    value_type: U_WORD
    register_count: 1

  - platform: modbus_controller
    modbus_controller_id: mppt

    name: MPPT energy
    address: 15217
    unit_of_measurement: "kWh"
    register_type: holding
    value_type: U_WORD
    register_count: 1

number:

  - platform: modbus_controller

    name: MPPT Float Volatge
    address: 10104
    value_type: U_WORD
    register_type: holding
    unit_of_measurement: "V"
    icon: mdi:timer-sand
    min_value: 24
    max_value: 56
    step: 1
    lambda: "return  x * 0.1; "

  - platform: modbus_controller
    id: mppt_max_current
    name: MPPT Corrente Max
    address: 10108
    value_type: U_WORD
    register_type: holding
    unit_of_measurement: "A"
    icon: mdi:timer-sand
    min_value: 1
    max_value: 80
    step: 1
    lambda: "return  x * 0.1; "
    write_lambda: |-
      ESP_LOGD("main","Modbus Number incoming value = %f",x);
      uint16_t b_capacity = x*10  ;
      payload.push_back(b_capacity);
      return x  ;

closed 2024-04-25 by jourdant

cobirnm, 2024-05-18:
Could you please share some pictures of how you did the conectou to ODU board?
What kind of connector did you use?
Thank you in advance and by the way excellent project!!!

jourdant, 2024-05-19:
Hi there @cobirnm - 

I took the row of pins from the ODU installer manual and wired up to how I currently have it deployed. The pins labelled Central Control A/B are the RS485 A/B pins. Those are the comms pins for the LGAP protocol on my ODU. I've also wired up the 12V/GND pins as that's important for the comms and if you're using the same M5Stack Atomic RS485 board I'm using, the 12V onboard the ODU can power the ESP32 too! The pins are even close together which is a bonus.

[image]

bzumik1, 2024-06-03:
Hi @jourdant,

I am still in the analysis phase (without the AC unit yet :) ). I would like to ask if, using LGAP and this integration, I can override the indoor temperature. I am asking because I would like to run the AC without the LG thermostat and provide the temperature from Home Assistant. I could measure it in multiple places to provide a more accurate value.

jourdant, 2024-06-03:
@bzumik1 thats a good question, it's something I'd like to figure out too. I haven't been able to figure out all the bytes in the protocol yet. From what we do know, I can't see a function for that yet.

It is a feature I've seen in some BMS deployments. That gives me hope that potentially it's possible. In theory even just disabling the internal thermostat would allow you to handle turning on/off the unit yourself with an automation based on an external temperature sensor.

bzumik1, 2024-06-03:
Thanks for the quick answer @jourdant. The bang-bang control is an option, but I believe the AC unit performs some optimizations (e.g., lowering the cooling when nearing the target temperature or increasing it if the difference is significant). I am surprised that this functionality is not supported by LG’s official MODBUS gateway.

I see that it is possible to achieve this using [this repository](https://github.com/JanM321/esphome-lg-controller), but I am more interested in your method of controlling the system. How complicated do you think it would be to identify the correct part of the communication protocol? How do you go about decoding it? Is there any documentation available?

bzumik1, 2024-06-03:
I have opened an issue for this feature https://github.com/jourdant/esphome-lgap/issues/8 so the discussion and progress can be tracked there.