jourdant/esphome-lgap #2: LG Therma V before 2019.

nedyarrd, 2024-03-02:
I have  THERMA V, ODU - HU071MR.U44 and IDU - HN0916M.NK4. For some time I try to connect them to my HA server but without any result. That IDU have onboard modbus port and even two dip for control it's behaviour. But still without success. I found in some documents on LG site that version of IDU have some issues and modbus communication doesnt work with that. So I have some questions about Your project:

- do You connect modbus directly to ODU or IDU?
- what voltage do You use, because I'm using EW11a wifi converter supplied by 5V, because at PCB is described as 5V
- in documentation that modbus is described as 9600-8N1 but it doesnt work at all as a MODBUS or LGAP

If You help me connect to my IDU or ODU i will try to help You to document LGAP protocol side.

With regards
Nedyarrd

jourdant, 2024-03-04:
Thanks for reaching out! You have a couple of options for integration. If you want to interface with the ODU then this project is likely for you, if you want to integrate with an IDU, then check this project out: https://github.com/JanM321/esphome-lg-controller.

I've pulled up the installation manual (https://gscs-b2c.lge.com/downloadFile?fileId=4qXYBRCQT7VWO8sw2Pg) and it seems like you have the following options for integration:
* There seems to be an onboard Modbus interface? at least it talks about it on a few pages. In the case where it may be possible to integrate with that interface, this video may be useful: https://www.youtube.com/embed/Xuj2YFZ5zME?si=3Ovf-9DifcNpapcV.
* Page 125 suggests you can potentially integrate an LG PI-485 board with your ODU. The PI-485 board will present a Central Control/LGAP interface that we can use with the protocol work I've been doing. Long story short - this option should work with the code I've been writing, but does require to purchase the PI-485 board. I've seen them on ebay in the past.

nedyarrd, 2024-03-04:
It doesn't matter what type of unit will I use, main goal is to at beginning monitor whole heatpump. JanM321 is one things I want to try, but Your project is more interesting, because:
- as You said I have RS485 interface onboard
- as I found in hn0916m.nk4 there is a problem with Modbus RTU (reference needed - I found one on ).
 -  NK5 is model where Modbus is working. 
 - https://www.intechna.cz/fotky81432/funkce/LG-Therma-V-AWHP-R32-Split-PDB.pdf - only one reference I found - for modbus there is need for RTU (module?) (page 42) in PDF
 - I think that CN_COM interface is a RS485 bus as it is stated in manual, but MODBUS function described on PCB is not working
 - maybe that interface can be used as described by You but protocol can be slightly different, I think it can be extended version like in JanM321 13 byte protocol. But I can't get it to work
 - I found that serviceman installed something like WiFi service stick to query information from ODU via something like RS485 for later service use.
 - other thing is that communication between ODU and IDU is by one additional wire it looks like it is some kind of RS485 too, but with common A or B wire connected to ground (GND)

Here are some photos of my PCB's 

[image]
[image]
[image]
[image]

So I have some questions:
- I thoutght Your project is to connect directly to ODU omitting PI-485 board? Do I need PI-485 then?
- it is possibile to run RS485 with 5V only? as it is seen on stickers on my PCB? Maybe that is my communication problem
- I think that when Modbus function is broken on this board it is possible that RS485 interface in IDU speaks some of LGAP protocol

Some additional findings:
- akb74855311 - display/controller uses that same protocol as described by JanM321
- akb74855311 - is linux PCB with TI - AM3354BZCZ60 ARM Cortex A8 processor
- have RS232 serial connector
- it looks like whole IDU, ODU and display shares that same data across multiple busses so no matter where we try to connect to desired data, so still main goal is to reverse engineer LGAP protocol

Additional photos of display PCB.
[image]
[image]
[image]
[image]

jourdant, 2024-03-24:
Apologies for the delay in my response. This is a lot of great information you've collected here. The photos are especially interesting. Just coming back to your questions:

>I thought Your project is to connect directly to ODU omitting PI-485 board? Do I need PI-485 then?
My project is supposed to work with any LGAP interface. In my particular case (Multi-V S), I have an LGAP interface on-board which is labelled as "Central Control A/B". I'll grab a photo of my PCB next time I open the unit. My understanding regarding the PI-485 board is where you have a unit that is older or doesn't have an RS485 interface, the PI-485 provides a bridge to these older devices. The PI-485 can speak both LGAP and non LGAP protocols depending on the DIP switches you select. 

>It is possible to run RS485 with 5V only? as it is seen on stickers on my PCB? Maybe that is my communication problem
I think that when Modbus function is broken on this board it is possible that RS485 interface in IDU speaks some of LGAP protocol
I haven't measured the voltage on my central control pins yet. I will do this and let you know next time I open the unit.

-

I also noticed in your images that you have some wires attached to the 3rd party controller A/B terminals with a resistor across them. Was that installed by you or by the installer that you mentioned? I re-watched the youtube video I had linked earlier to check the label that's on his terminals and it actually matches the same as what you're seeing on yours. That does make me suspect that this is actually a modbus interface rather than LGAP especially given his unit was also a Therma V unit.

Just to be clear, are those the main integration pins you're observing? or are you also looking at some other pins as potentially carrying the LGAP signal?

To your point around IDUs also potentially using LGAP protocol, I do think you're right. There are also indoor specific SKUs of the PI-485 also.

nedyarrd, 2024-04-16:
No proble, spring is in way, so I have many work outside house ;-) I didn't look here either.
> Apologies for the delay in my response. This is a lot of great information you've collected here. The photos are especially interesting. Just coming back to your questions:
>
>
> My project is supposed [...]
Yeah I attached wires to simply terminate RS485 line. It didn't helped anyway. 
> I also noticed in your images that you have some wires attached to the 3rd party controller A/B terminals with a resistor across them. Was that installed by you or by the installer that you [...]
On dip switch there are two settings:
1st - master - slave
2nd - Exclusive/General (protocol selection)

That same connector is used for ie. Power meter interface. 

I did try some other things with this connector, but I might have a RS485 problem. Because sometimes I have responses that I don't see. So at beginning:

I wrote python script to use my EW11A modbus/WiFi converter it does:
- change baud rate
- change parity/bits etc 
- send sequences of bytes:

I found that:
- correct way to connect modbus is A->B B->A not A->A and B->B to have something.
- controller at startup send pack of information at slow speed, it is slower than 300bps, because at 300bps it is 0xFF
- any response from controller is not full real time, sometimes it takes some seconds, because after some combinations ie. 9600-8N1 4800-8N1 non typical settings like 2400 or 9600-8E2 - have some weird responses in about 23 bytes length. Sadly I don't know what is in response because EW11A filters them out :( And shows only statistics.

> Just to be clear, are those the main integration pins you're observing? or are you also looking at some other pins as potentially carrying the LGAP signal?

Yep there are some more pins, that I think can be useful.
> 
> To your point around IDUs also potentially using LGAP protocol, I do think you're right. There are also indoor specific SKUs of the PI-485 also.

jourdant, 2024-04-18:
There's some interesting insights there. I've made some pretty significant progress since my last update too.

I found that a number of people were also having issues with the LILYGO T-RSC3 and the manufacturer has been silent on the subject. With that in mind I decided to try a different board (M5Stack Atom Lite)[https://shop.m5stack.com/products/atom-lite-esp32-development-kit] (which I had on-hand) and I added on (ATOMIC RS485 Base)[https://shop.m5stack.com/products/atomic-rs485-base]. 

This setup uses a different RS485 chip that doesn't require a flow control pin and the other nice thing is that it has 4 pins in a row - GND, 12V, A, B and can power the board using the 12V pin. That's perfect for my use case plugging into my outdoor unit as I have an internal 12V and GND pin right next to the central control pins.

[image]

After putting everything together, changing the target in esphome to use this the Atom Lite instead, the custom driver I wrote worked! I was able to send requests and get bytes back, decode the bytes and enable the Climate entities to show up within HA. I'm now working on optimising the driver as there are a couple of timing issues, but my main worry of comms is all resolved.

To your point about finding data at very slow speeds, JanM321's project speaks the wall controller protocol at 104bps, perhaps that's what you're seeing there?

MaikaiLife, 2024-11-21:
> There's some interesting insights there. I've made some pretty significant progress since my last update too.
> 
> I found that a number of people were also having issues with the LILYGO T-RSC3 and the manufacturer has been silent on the subject. With that in mind I decided to try a different board [M5Stack Atom Lite](https://shop.m5stack.com/products/atom-lite-esp32-development-kit) (which I had on-hand) and I added on [ATOMIC RS485 Base](https://shop.m5stack.com/products/atomic-rs485-base).
> 
> This setup uses a different RS485 chip that doesn't require a flow control pin and the other nice thing is that it has 4 pins in a row - GND, 12V, A, B and can power the board using the 12V pin. That's perfect for my use case plugging into my outdoor unit as I have an internal 12V and GND pin right next to the central control pins.
> 
> ![IMG_4049](https://private-user-images.githubusercontent.com/230794/323773928-f7eeaa1f-5c5c-4174-b05f-4a951c122593.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIxNzc4MzUsIm5iZiI6MTczMjE3NzUzNSwicGF0aCI6Ii8yMzA3OTQvMzIzNzczOTI4LWY3ZWVhYTFmLTVjNWMtNDE3NC1iMDVmLTRhOTUxYzEyMjU5My5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyMVQwODI1MzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iYzI1MWUwZDA0OTQ1NmIwM2NiOTY0MGIyNTIxYmIwNmE4ZjQ4ZGNlNjM2YjRhNjYyMzEyNjAwYjkwMGZiOGQ0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.tQkuPSzim8roGs110jKS9jeoAWdatl5-cEYkkYG2nPg)
> 
> After putting everything together, changing the target in esphome to use this the Atom Lite instead, the custom driver I wrote worked! I was able to send requests and get bytes back, decode the bytes and enable the Climate entities to show up within HA. I'm now working on optimising the driver as there are a couple of timing issues, but my main worry of comms is all resolved.
> 
> To your point about finding data at very slow speeds, JanM321's project speaks the wall controller protocol at 104bps, perhaps that's what you're seeing there?

I have the same RS45 board, but with an Atom Matrix display.  I'm still stuck as you might notice from all my comments at the getting communication happening over RS485.  I have the PI PMNFP14A1, with dips 1 and 4 enabled, but the green comm light never blinks, I've tired all the suggestions here, just not sure where else to go, and I don't really have other options but to get this working.

Happy to send photos, give debug logs, just wondering why I can't seem to see any communication with the PI?

Here is what my setup looks like.

[image]

nedyarrd, 2026-06-27:
Reopenning that thread because:
- serviceman from my heatpump said that int that type of hydrosplit modbus port is broken
- I proved above by metering voltage and resistance on port, it seems good, but without any signal from start of heatpump
- on Chinese LG site I found information that with that model modbus is onboard but is not supported because of firmware failure
- modbus on that board is very prone of voltage drift of line A :) that came from experiments, so to have any echo of something You need ground to PCB, but there is still no reply from firmware
- so I tired to build esp-lgap from [JAN](https://github.com/JanM321/esphome-lg-controller) and it worked for me, 

And there are My findings about that protocol for lgap panels to LG Therma Heatpumps:

- protocol is not 13 bytes long, but is 20 bytes long
- protocol speed is not 104bps, for that panels is 300bps/N/1
- checksum is that like in JANM321 project
- now I can access almost every piece of setting/sensor what is available on panel (some things need to be still reverse engeenered)
- probably WiFi port communication has that same standards but fever info
- on ODU unit there is a service port that probably has that same protocol and settings but more info about  ODU, like temperatures, pressures, fan speed, and compressor speed

On august I will try to publish my findings with esphome example for my case. By now I can read these sensors:
- DHT tank setpoint/current temperature
- DHT tank 3 way valve state
- Thermostat state
- Info about cooling/heating mode
- Internal pump flow value
- Setpoint for cooling/heating
- Silent mode setting
- Panel temperature

apszowski, 2026-09-11:
> Reopenning that thread because:
>
> * serviceman from my heatpump said that int that type of hydrosplit modbus port is broken
> * I proved above by metering voltage and resistance on port, it seems good, [...]

Hi @nedyarrd 
did you had to modify any part of the code from [JAN](https://github.com/JanM321/esphome-lg-controller)  to get it working with a Therma V Pump ? I'm also struggling with my Therma V NK3 to get MODBUS working, so I'm looking for alternative options.