jourdant/esphome-lgap #8: Indoor Temperature Readings Override

bzumik1, 2024-06-03:
I am opening this issue to track the progress of this feature. It should allow you to override the room temperature. This will enable you to get rid of the LG thermostat and combine the temperature reading from multiple sensors for better accuracy.

This functionality is available as part of this integration [this repository](https://github.com/JanM321/esphome-lg-controller) but based on the information we have, it should be possible to add it using LGAP as well.

MaikaiLife, 2024-12-05:
> I am opening this issue to track the progress of this feature. It should allow you to override the room temperature. This will enable you to get rid of the LG thermostat and combine the temperature [...]

@bzumik1 

Did you ever get this working, myself and a couple of others would like to keep this project going but it seems everyone initially involved has dropped off since about June of 2024.

mystardious, 2025-11-27:
I don't think this is possible since what it does is change the reported room temperature by the LG Controller back to the ODU and AFAIK isn't supported.