jourdant/esphome-lgap #10: PI485 LGAP +  ATOM Lite & ATOMIC RS485 BASE not communicating with the PI485:  need help with the settings

TerryFrench, 2024-07-20:
TL;DR: I'm looking for help with my uart config - most likely

Hi Jourdan,
Your project is awesome, and I picked some LG heat pump based on the availability of your code :-) (and the price vs. Mitsubishi I also control using esphome). I'm DIY, so self install. outdoor unit: A4UW30GFH0 = LMU300HHV in the USA. and 3 indoor units:  two AMNW12GQA0  = LQN120HV4 and one AMNW09GQA0 = LQN090HV4. 
All units work fine using the IR remote controls. 

I installed a PI-485 board today, and the dipswitch settings should be fine here - and are set to enable the LGAP protocol. 
I connected the Atom-RS485 gnd to gnd on the PI-485, 12V to the 10V on the PI-285 (about 10.8 V really), A to A on the Atom, and B to B on the Atom.
Note: I have not twisted A&B wires yet as I just found out they should be twisted so there might be some 60Hz noise at least there given the proximity to the mains' power wires. The wires length is about 15 cm / 6 inches. 

I do **not** think I have the uart settings correct, so I don't see any real data going. I also tried changing the settings in homeassistant to set the temperature... but while the settings are sent, likely nothing really happens. 

I tried various pins, although TX on GPIO22 and RX on GPIO19 seem that they should be ok. 
I tried 9600 bauds and 115200 bauds, but only get some warnings that it's not compatible with the integration, no differences. 
I tried using "inverted: true" for the TX and RX pins, no difference. 
I added dummy_receiver: true to the uart debug section, but saw no data displayed. 

The PI-485 green led (RS-485 data) is steady / not blinking, so likely not getting data it can recognize.

Relevant logs examples showing nothing is received.  
[01:38:21][C][logger:186]:   Level: DEBUG
[01:38:21][C][logger:188]:   Log Baud Rate: 0
[01:38:21][C][logger:189]:   Hardware UART: UART0
[01:38:21][C][uart.arduino_esp32:151]: UART Bus 0:
[01:38:21][C][uart.arduino_esp32:152]:   TX Pin: GPIO22
[01:38:21][C][uart.arduino_esp32:153]:   RX Pin: GPIO19
[01:38:21][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[01:38:21][C][uart.arduino_esp32:157]:   Baud Rate: 4800 baud
[01:38:21][C][uart.arduino_esp32:158]:   Data Bits: 8
[01:38:21][C][uart.arduino_esp32:159]:   Parity: NONE
[01:38:21][C][uart.arduino_esp32:160]:   Stop bits: 1
[01:38:21][C][uptime.sensor:034]:   Type: Seconds
[01:38:21][C][lgap:016]: LGAP:
[01:38:21][C][lgap:019]:   Flow Control Pin:
[01:38:21][C][lgap:026]: Flow control pin not set.
[01:38:21][C][lgap:029]:   Loop wait time: 500ms
[01:38:21][C][lgap:030]:   Receive wait time: 4000ms
[01:38:21][C][lgap:031]:   Child devices: 3
[01:38:21][C][lgap:034]:   Debug: true
[01:38:21][C][lgap.climate:018]: LGAP HVAC:
[01:38:21][C][lgap.climate:019]:   Zone Number: 0
[01:38:21][C][lgap.climate:020]:   Mode: 0
[01:38:21][C][lgap.climate:021]:   Swing: 0
[01:38:21][C][lgap.climate:022]:   Temperature: 0
[01:38:21][C][lgap.climate:018]: LGAP HVAC:
[01:38:21][C][lgap.climate:019]:   Zone Number: 1
[01:38:21][C][lgap.climate:020]:   Mode: 0
[01:38:21][C][lgap.climate:021]:   Swing: 0
[01:38:21][C][lgap.climate:022]:   Temperature: 0
[01:38:21][C][lgap.climate:018]: LGAP HVAC:
[01:38:21][C][lgap.climate:019]:   Zone Number: 2
[01:38:21][C][lgap.climate:020]:   Mode: 0
[01:38:21][C][lgap.climate:021]:   Swing: 0
[01:38:21][C][lgap.climate:022]:   Temperature: 0
...
[01:38:21][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[01:38:21][D][lgap.climate:221]: Generating READ request message for zone 0...
[01:38:21][D][uart_debug:114]: >>> 00:00:FA:00:00:00:F1:BE
[01:38:25][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[01:38:25][D][lgap.climate:221]: Generating READ request message for zone 1...
[01:38:25][D][uart_debug:114]: >>> 00:00:FA:01:00:00:F1:B9
[01:38:29][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[01:38:29][D][lgap.climate:221]: Generating READ request message for zone 2...
[01:38:29][D][uart_debug:114]: >>> 00:00:FA:02:00:00:F1:B8
[01:38:33][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[01:38:33][D][lgap.climate:221]: Generating READ request message for zone 0...
[01:38:33][D][uart_debug:114]: >>> 00:00:FA:00:00:00:F1:BE
[01:38:37][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[01:38:37][D][lgap.climate:221]: Generating READ request message for zone 1...
[01:38:37][D][uart_debug:114]: >>> 00:00:FA:01:00:00:F1:B9
[01:38:41][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[01:38:41][D][lgap.climate:221]: Generating READ request message for zone 2...
[01:38:41][D][uart_debug:114]: >>> 00:00:FA:02:00:00:F1:B8
[01:38:45][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[01:38:45][D][lgap.climate:221]: Generating READ request message for zone 0...
[01:38:45][D][uart_debug:114]: >>> 00:00:FA:00:00:00:F1:BE
[01:38:49][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[01:38:49][D][lgap.climate:221]: Generating READ request message for zone 1...
[01:38:50][D][uart_debug:114]: >>> 00:00:FA:01:00:00:F1:B9
[01:38:53][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[01:38:53][D][lgap.climate:221]: Generating READ request message for zone 2...
[01:38:54][D][uart_debug:114]: >>> 00:00:FA:02:00:00:F1:B8
[01:38:57][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[01:38:57][D][lgap.climate:221]: Generating READ request message for zone 0...
[01:38:58][D][uart_debug:114]: >>> 00:00:FA:00:00:00:F1:BE

yaml file (renamed .txt for github)
[lg-lmu300hhv.yaml.txt](https://github.com/user-attachments/files/16319516/lg-lmu300hhv.yaml.txt)

Thanks in advance for any help you, or someone else might provide!

kkrankall, 2024-07-20:
I'm will be starting my install this week. Have a multi F max with PI-485 board. I had the same questions about the uart settings. Good luck let m know how it goes.

TerryFrench, 2024-07-20:
> I'm will be starting my install this week. Have a multi F max with PI-485 board. I had the same questions about the uart settings. Good luck let m know how it goes.

I will post any updates.  My is the Multi-F "not max" :-) i.e. 4 sets of pipes exiting the ODU instead of 1 set like yours (likely), and the Branch distribution unit(s) you are likely using with it. 

I might get another RS-485 to Wifi/TCP device I use for a solar controller (but I hate changing its settings, it was a bear to get working...) and send some of the query packets that should be good based on the protocol and logs I have to see if I get any response, and what settings seem to work. No esphome/homeassistant involved.

TerryFrench, 2024-07-22:
small update: still no success. Although I can confirm 4800bps should be the right speed. 
I'll receive a USB-RS485 adapter today, hoping to find some time this week to use it (PC far from the ODU, need to borrow wife's laptop). 
I tried swapping A/B, no differences, back to proper connection. 
I did twist the A&B cables, and also added some 120 Ohm resistors across A&B at both ends (tests with all powered off had high resistance, so not terminated and no jumper to terminate either. 

I see pretty much no differential voltage with a multimeter, but that's likely because the bus is set in slave on the PI-485 side, and I likely do not send / switch to send mode on the M5-stack side. - or at least I think that's one hypothesis.

kkrankall, 2024-07-30:
Terry, how are you making out?

TerryFrench, 2024-07-30:
I connected my digital oscilloscope but did not manage to capture any signal, even with a low trigger. 
So I temporarily gave up on the outdoor unit - I will have to try RS-485 between 2 devices first. 
The PI-485 has the most left led (orange) blinking non stop, and overall I believe that the board is not happy the way it is set with the motherboard. If I had to guess, I'd say the firmware on the MB might not have the functionalities turned on to communicate with the Pi-485, being a "LG Multi F" (residential) but not "LG Multi F Max" (more for apartment complex...) 
I did try multiple dip switches settings, even some that should not be relevant to my model, but the model names for the PI-485 and my outdoor model are really different/not really matching.

Note that the motherboard has a RS-485 port that has no connector soldered in, but I wouldn't bet on the firmware to work well. 

Anyway, I ordered 20 PCB boards for the indoor units to be made from the design on https://github.com/JanM321/esphome-lg-controller  and I should receive them today :-) 
So I hope to try that out tonight, and if that works, keep 5 and sell the rest more or less at cost + shipping (+gas to get to the post office, but you get the idea: super cheap.)

I do really like the idea of communicating with the external unit, but I'm disappointed in the PI-485 documentation from LG. (ok, I really do Not like any doc from LG vs. Mitsubishi that lets you access their docs). 

I might buy a cheap logic analyzer and use that as a better / more sure way to analyze what is going on with the output of the PI-485, but really I will need to build a cable to put in between the motherboard and the PI-485 board to see if they are at least communicating with each-other. 
Last ditch: I will buy one of the LG troubleshooting tools (LG Sims I think, or the other one), to access all sensor values and other settings once in a while, because I think maintenance-wise, that'd be pretty neat to monitor how the unit is doing. 

Note: for the Mitsubishi, I use https://github.com/unixko/MitsuCon

ChriD, 2024-07-31:
@TerryFrench 
If those are the "tiny" designs i may want to get them from you :-)

TerryFrench, 2024-07-31:
> @TerryFrench If those are the "tiny" designs i may want to get them from you :-)

tiny yes. Not received yet (DHL delays).

kkrankall, 2024-08-01:
I bought 5 pcb and have 4 connected. They work great. Wanted this project to work as I have two wall units where the remote ports are fried and get a CH03 error when hooking up a wired remote. I dont feel like replacing the main PCB and wanted local control.

TerryFrench, 2024-08-02:
The boards work well. 
I have 3 indoor units ("wall console" type), built 3 cables, and had no issues. 
esphome config is as https://github.com/JanM321/esphome-lg-controller has it. 
I'll have to dig what the fan settings mean, as I changed the values (100% seems to be the full speed with the natural breeze mode) 
And the vanes direction is opposite (close vs open as my unit is low on the wall and LG decided something not compatible ;-) 
Initial connection showed some CRC errors for one unit, but they have since disappeared. 

I've included photos of the boards and a rough installation location. My new 3D printer is not yet functional (Voron 2.4, built, but I have some SW config to fix), so I used electrician tape at the bottom, then cardboard, and hot glue to get all that in. 

I still want to get the ODU RS-485 to work, but getting some form of remote control was my first priority, so I can get some rooms at the right temperature before coming from work. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

TerryFrench, 2024-08-02:
> @TerryFrench If those are the "tiny" designs i may want to get them from you :-)

Same reply as here https://github.com/JanM321/esphome-lg-controller/issues/31#issuecomment-2264593521 
How many boards, and to what country? (so I can figure out the shipping cost) 
These are just the boards and 1 cable. I don't have more ESP boards to go with them, but they are easy to buy from China or other places like Amazon in most countries, I believe.

ChriD, 2024-08-02:
@TerryFrench 
Looks great!
I'd take 6 of them and the Shipping would be to Austria (not to be confused with Australia)

ChriD, 2024-08-09:
@TerryFrench 
Have you had time to check the shipping costs?

TerryFrench, 2024-08-09:
sorry not yet.
Price will be $15 per board + shipping

ChriD, 2024-08-18:
@TerryFrench 
If you want to sell,. I 'l be there waiting ;-)

MaikaiLife, 2024-11-21:
> TL;DR: I'm looking for help with my uart config - most likely
>
> Hi Jourdan, Your project is awesome, and I picked some LG heat pump based on the availability of your code :-) (and the price vs. [...]

Did you ever get this to work, I have the pi485 with an LG Multi F and I never see the green light blink?

greyburn, 2024-11-26:
I'm in a similar position, PI-485. I got the green light to blink, but not receiving any data. Going to get a USB receiver to test with a computer.
Anyone have a working config for PI-485?

MaikaiLife, 2024-11-26:
What pin settings did you use to get the green light to blink? I'm wondering if you need to power cycle the whole AC unit.

greyburn, 2024-11-26:
1&4 on, all others off.
I've power cycled a few times. Swapped A&B, changed baud (back to 4800 now). No joy :(

MaikaiLife, 2024-11-26:
Okay, I've tried those same things also.  I've also read on this page that when you boot the PI it will blink the number of units it detects.  Now my unit is a Multi-F, model LMU480HV with 4 inside ceiling cartridges.

The instructions are for the Multi-V but there is some similarity of how these things work.

[image]

I wish I had known this was so far off from working before I mounted everything.

Here is what my install looks like, with a blank for the other unit when and if I get it working.  I have 2x of the LMU480HV units each with 4 cartridges.

[image]

I'll keep posting my progress here, I've dropped $400 on PI's and ESP devices, boxes, etc so I'm pretty determined to try and figure something out.

greyburn, 2024-11-26:
Good to know I'm not alone :)
I've never got the red light to blink twice (I have two indoor units). I've ignored that for now. But maybe I have not addressed my indoor units correctly. I really don't want to have to open them up :( :( :( I get one flash, I don't see why that would prevent RS-485 from working though but LG hasn't exactly shared a lot of documentation on all this.
Have you tried a computer with a RS-485 interface?

MaikaiLife, 2024-11-26:
> Good to know I'm not alone :) I've never got the red light to blink twice (I have two indoor units). I've ignored that for now. But maybe I have not addressed my indoor units correctly. I really [...]

I haven't tried to use a computer yet, I'm starting to wonder if the stack on the ESP device even speaks any RS485, I was actually gonna try and make the two units I have talk together to confirm they actually work. I have the  following components.

https://shop.m5stack.com/products/atomic-rs485-base

https://shop.m5stack.com/products/atom-matrix-esp32-development-kit

I looked all over the internet trying to figure out how to give addresses to the units, thinking maybe that's the issue, but I haven't yet applied it.

This YouTube video with no English is the only thing I could find that shows you how to use the remote to do the device ID assigment.
https://youtu.be/PRSRoJBTqSI?si=RWNE7yuQ-BLsROUJ

greyburn, 2024-11-26:
Thanks for the addressing link - I'll try this in a few hours.
I'm using the exact same hardware as you. I too am wondering if the M5stack or the Pi485 is not speaking RS485. I figure a laptop with a USB RS-485 adapter will allow me to test both and see which one is misbehaving. I should have an adapter in a week.

greyburn, 2024-11-26:
OK I had a minute, so programmed my ID's on the indoor units (01 & 02). The PI-485 is reporting 3 zones now (not true, only 2 but the unit is 3 capable). I think that's ok. No initial luck, but i need to swap A&B around and check RX/TX pins (the Atom documentation has them reversed, but physically its written as RX22 TX19 on the device so not sure which is correct. At least the pi-485 seems to be behaving as expected now so hopefully its just A/B wiring now. I think I am close!

MaikaiLife, 2024-11-26:
Here’s my ESPHome configuration—glad to hear the IDs had some effect with the lights blinking. My initial thought was that it might not send any data over RS485 if no zones were detected.

[AC-Compressor1.txt](https://github.com/user-attachments/files/17925460/AC-Compressor1.txt)

greyburn, 2024-11-27:
Thanks, our config is the same pretty much. 
I played a little with options but no luck. I think I'm close now. Likely just need to go through every option and possibly power cycle after every change. That will be time consuming!
You should set your IDs and see if you have any success. Setting the ID was easy, I followed the video. It didn't show what to do after but I just power cycled the remote and then pressed the power button (prior the LED display was just showing the ID).

MaikaiLife, 2024-11-28:
I'm gonna give it another go in the AM tomorrow, I'll let you know what I discover if you have any breakthroughs @greyburn please be sure and share, we're in this together, cause I'm not sure anyone else is watching this repo anymore. :)

MaikaiLife, 2024-11-28:
Here’s an update on my progress:  

I’ve programmed the IDs on all four of my indoor units. When I booted the Pi, the red light blinked four times, indicating it detected the correct number of indoor units. That’s definitely an improvement! However, I’m still not seeing a green light blink.  

I’ve tried swapping pins—both physically and virtually—but the Pi and ESP device still don’t seem to communicate. I have another base unit, the Atom Lite, which I’ll bring back with me on December 20th to swap with the Matrix and see if that makes a difference.  

In the meantime, I’ll continue experimenting with the pins, configuration libraries, and other settings to see if I can make further progress.  

Stay tuned!

MaikaiLife, 2024-11-28:
More attempts at progress.

1. I swapped PI's since I have two of them, no difference.
2. I changed power supplies to USBC for testing, no difference.
3. I changed RS485 Atom boards, no difference.
4. I swapped A & B wires, no difference.

So far, I think the PI is talking to the indoor units, at least it's detecting them and blinking like it's talking, but I've yet to be able to get a response from the PI over RS485.

Hopefully, the original author of this repo @jourdant  might chime in and help so we can confirm the PI working and really make this project fully compatible with LG equipment that requires the PI. 🤞🙏

greyburn, 2024-11-29:
Thanks for sharing progress, I'll continue to do the same.
I should have a USB-<>RS-485 today and will test that with the pi-485 to see if that's at least working.

greyburn, 2024-11-29:
One thing I have noticed is https://www.lg.com/gr/business/download/airsolution/Gateway_Leaflet[20210524_185214613].pdf suggests we need to use a 9600 baud. But if that's a problem, expecting the computer connected to verify what speed/settings we need.

ASI-Kevin, 2024-12-02:
I'm interested to see what you find out. I kind of gave up on this project. Good luck.

greyburn, 2024-12-04:
My RS485 USB adapter finally arrived. I didn't get much time to play with it but had no success. I tried 4800 and 9600 baud connections but didn't get anything on terminal. I thought I had the green light flashing before but it is not, I may have been mistaken. I do see the Tx light flash on my adapter when I send data to the PI485, but I get no RX back. I tried rebooting the adapter a few times and swapping a/b but no luck :( I need some more time to try a few other things but need to wait for it to not be 30F outside! Hopefully later this week, Saturday at the latest.

greyburn, 2024-12-04:
had a little more time so I connected my esp (m5stack) to the adapter and I can see the read requests, so I know that is working. Starting to feel a little stumped, not sure what the pi-485 is expecting :( im going to connect it back up tonight and turn some units on/off and see if it writes anything to the port. 
Any luck on your end @MaikaiLife ?

MaikaiLife, 2024-12-04:
> had a little more time so I connected my esp (m5stack) to the adapter and I can see the read requests, so I know that is working. Starting to feel a little stumped, not sure what the pi-485 is [...]

Thank you for the update, I will be fiddling with my unit on the 21st, I really would like to get this going.  I'm starting to wonder if we're going to need some sort of device such as this RS485 to ethernet speaks the RS485 stuff so we can see what it's sending to ger response from the pi.

Watch this video to see if there are any ideas or tips that help understand how to chat with the PI.

https://www.youtube.com/watch?v=Xuj2YFZ5zME&t=794s&ab_channel=RodMcBain

MaikaiLife, 2024-12-04:
I'm thinking something like this would send the RS485 commands and maybe there is a way to listen and see them.

https://www.waveshare.com/rs485-to-wifi-eth.htm

greyburn, 2024-12-05:
Interesting Rod McBain also had 9600 baud configured. I also found in protocol.md that "To my knowledge right now, the ODU does not voluntarily send state updates. Instead values must be polled from the ODU by sending 8 byte requests."

I think I'm going to either build a new component using local files and 9600 configured, OR figure out a few valid commands and send them from my laptop using my adapter set to 9600. The first is probably easier.

MaikaiLife, 2024-12-05:
Interesting update, unlike the other project where the boards connect to each indoor unit, those send their status updates repeatedly.  However, you're correct I read the same thing, unless you query the PI for the status of a zone, you won't get an update on that zone.    I downloaded the repo and started looking at it, I've done some Python but no C++ so... I'm at least oozing through some steps to try and make modifications also.   It's too bad @jourdant who worked on this project as recently as June of 2024 seems to be gone, it would great to have his trail and error assistance to give us a little nudge in the right direction.

MaikaiLife, 2024-12-05:
Just a little more info on the PI485, it seems that at least this source references 4800.

https://doc.ipesoft.com/display/D2DOCV23EN/LG+PI-485+protocol

greyburn, 2024-12-05:
Interesting! I'm starting to wonder if the pi-485 actually speaks lgap and isn't just simple rs-485?

I created a local rep of lgap and changed check_uart_settings to 9600, but after checking the documentation I'm pretty sure this is not a problem as all it does it print an warning : "Check that the configuration of the UART bus matches the provided values and otherwise print a warning." so I don't think this is a problem :(

MaikaiLife, 2024-12-05:
Possibly….  but the 4th pin from the left according to the PI divination turns LGAP on/off so maybe you can turn it off and talk straight RS-485.

greyburn, 2024-12-05:
Reinstalled my m5stack with local code and check_uart_settings(4800) changed to 9600. No change, as expected.
It's still odd how the PI485 doesn't flash the green light. I'm running out of ideas :(

MaikaiLife, 2024-12-10:
@greyburn any more progress?  I'm not going to be in front of my unit until the 21st, or 23rd.  Just curious what you've already experimented with?

greyburn, 2025-01-26:
I more or less paused efforts here - although I did pick up a waveshare device to play with, I figure it will help determine where the comms issue is at least. But it's been so cold here that working on this has been impossible!
Hoping for better weather soon (its overdue) so before I go back to this wanted to check @MaikaiLife did you have any success?

MaikaiLife, 2025-01-28:
Hi @greyburn , I hate to say it but I gave up.  I instead purchased 8 of the little boards to put in the indoor units and they've been working fine for the last month now.

MaikaiLife, 2025-01-28:
I ordered the boards that are up further in this chat thread.

cobirnm, 2025-02-02:
I'm considering acquiring a PI485 board. Did anyone had success with this. Unfortunately my indoor units do not have the ability to connect single boards so my only hope would be to control using my outdoor units MU4M25 with a PI485 boardd connected to it, if I understand right.

Has anyone tried to connect the m5sack directly to CN_central in the outdoor PCB?

greyburn, 2025-02-07:
I'm not aware of anyone who has got the PI485 to work. I tried as did others above to no success.

cobirnm, 2025-02-13:
I've found this statement in the following link, maybe this helps...

https://descargas.futurasmus-knxgroup.org/doc/en/intesis/19697/intesisbox_lg-ac-mbs-4-8-32-64_installation_manual_en.pdf

[image]

Anoxym, 2025-02-13:
Intensis module works with ac unit to send and receive commands, I have one in my setup (Intesis LG-RC-MBS-1), linked to the [ESP32-S3-6CH-RELAY ](https://www.waveshare.com/wiki/ESP32-S3-Relay-6CH 1)

[image]

Here is the ESPHome code I use : 

`esphome:
  name: "lg-ac"
  friendly_name: "LG AC Controller"
  platform: esp32
  board: esp32-s3-devkitc-1

# Enable logging
logger:
  level: DEBUG
  # level: DEBUG

uart:
  id: uartx
  tx_pin: GPIO17  # TX pin ESP32 board
  rx_pin: GPIO18  # RX pin  ESP32 board
  baud_rate: 9600
  parity: NONE
  stop_bits: 2

modbus:
  id: modbus1
  uart_id: uartx

modbus_controller:
  - id: modbus_device
    address: 0x20   ## address of the Modbus slave device on the bus
    modbus_id: modbus1
    update_interval: 3s

################################# CONTROL ################################
switch:
  - platform: modbus_controller
    modbus_controller_id: modbus_device
    name: "AC On/Off"
    id: ac_on_off
    register_type: holding
    address: 0x0000
    lambda: |-
      if (x) {
        return 1;  // Turn ON
      } else {
        return 0;  // Turn OFF
      }

number: 
  - platform: modbus_controller
    modbus_controller_id: modbus_device
    name: "Mode"
    id: ac_mode_set
    register_type: holding
    address: 0x0001
    value_type: U_WORD
    step: 1
    min_value: 0
    max_value: 4
    icon: "mdi:auto-mode"

  - platform: modbus_controller
    modbus_controller_id: modbus_device
    name: "Fan speed"
    id: fan_speed_set
    register_type: holding
    address: 0x0002
    value_type: U_WORD
    step: 1
    min_value: 0
    max_value: 7
    icon: "mdi:fan-plus"

  - platform: modbus_controller
    modbus_controller_id: modbus_device
    name: "Temperature setpoint"
    id: temperature_setpoint_set
    register_type: holding
    address: 0x0004
    value_type: U_WORD
    step: 0.5
    min_value: 160
    max_value: 300
    mode: box
    icon: "mdi:temperature-celsius"

############################## SHOW SENSORS ##############################
text_sensor:
  - platform: modbus_controller
    modbus_controller_id: modbus_device
    id: ac_mode_show
    register_type: holding
    address: 0x0001
    raw_encode: HEXBYTES
    name: "Mode"
    icon: "mdi:auto-mode"
    lambda: |-
      uint16_t value = modbus_controller::word_from_hex_str(x, 0);
      switch (value) {
        case 0: return std::string("Auto");
        case 1: return std::string("Heating");
        case 2: return std::string("Dehumidification");
        case 3: return std::string("Fan");
        case 4: return std::string("AC");
        default: return std::string("Unknown");
      }
      return x;

  - platform: modbus_controller
    modbus_controller_id: modbus_device
    id: fan_speed_show
    register_type: holding
    address: 0x0002
    raw_encode: HEXBYTES
    name: "Fan speed"
    icon: "mdi:fan"
    lambda: |-
      uint16_t value = modbus_controller::word_from_hex_str(x, 0);
      switch (value) {
        case 0: return std::string("Auto");
        case 1: return std::string("Position 1");
        case 2: return std::string("Position 2");
        case 3: return std::string("Position 3");
        default: return std::string("Unknown");
      }
      return x;

  - platform: modbus_controller
    modbus_controller_id: modbus_device
    id: status_compressor_show
    register_type: holding
    address: 0x0035
    raw_encode: HEXBYTES
    name: "Compressor status"
    icon: "mdi:engine"
    lambda: |-
      uint16_t value = modbus_controller::word_from_hex_str(x, 0);
      switch (value) {
        case 0: return std::string("Off");
        case 1: return std::string("To off");
        case 2: return std::string("To on");
        case 3: return std::string("On");
        default: return std::string("Unknown");
      }
      return x;

      
sensor:
  - platform: modbus_controller
    modbus_controller_id: modbus_device
    name: "Temperature setpoint"
    id: temperature_setpoint_show
    register_type: holding
    address: 0x0004
    value_type: U_WORD
    unit_of_measurement: "°C"
    icon: "mdi:temperature-celsius"

  - platform: modbus_controller
    modbus_controller_id: modbus_device
    name: "Temperature reference"
    id: temperature_reference
    register_type: holding
    address: 0x0005
    value_type: U_WORD
    unit_of_measurement: "°C"
    icon: "mdi:temperature-celsius"

  - platform: modbus_controller
    modbus_controller_id: modbus_device
    name: "AC time"
    id: ac_unit_op_time
    register_type: holding
    address: 0x0009
    value_type: U_WORD
    unit_of_measurement: "h"
    icon: "mdi:timer-sand"

  - platform: modbus_controller
    modbus_controller_id: modbus_device
    name: "Compressor time"
    id: compressor_op_time
    register_type: holding
    address: 0x0036
    value_type: U_WORD
    unit_of_measurement: "h"
    icon: "mdi:timer-sand"

  - platform: modbus_controller
    modbus_controller_id: modbus_device
    name: "Alarm status"
    id: alarme_status
    register_type: holding
    address: 0xA
    value_type: U_WORD
    icon: "mdi:alert-circle"

  - platform: modbus_controller
    modbus_controller_id: modbus_device
    name: "Error code"
    id: error_code
    register_type: holding
    address: 0xB
    value_type: U_WORD
    icon: "mdi:alert-circle"`

greyburn, 2025-02-13:
Oh interesting re termination resistor. I still have my pi485 hooked up so I'll give this a go.

cobirnm, 2025-02-24:
> Oh interesting re termination resistor. I still have my pi485 hooked up so I'll give this a go.

Did you have any success with this?

greyburn, 2025-03-09:
I tried the termination resistor (120 ohm) with the pi-485 : no luck. 
I'm not sure the pi-485 is interfacing correctly but im giving up and switching to the [esphome-lg-controller](https://github.com/JanM321/esphome-lg-controller) project.

jourdant, 2025-05-12:
@greyburn @cobirnm  sorry both - I had things come up that took my attention away from this project.  I understand you may have moved to the esphome-lg-controller project (which works really well btw!). If there's still interest in this, happy to start a new thread and take a look. Also interestingly, @szolnokit uploaded a scanned copy of the spec which I've also merged in. I'll look to update what I have according to this document https://github.com/jourdant/esphome-lgap/blob/e55d422305c4d41485d46adde522608ef3296421/images/LGAP.pdf.

kkrankall, 2025-05-12:
Hi @jourdant ,

Thank you for the update and for all the time you've dedicated to this project. I'm still very interested in continuing with this and appreciate the effort you've put in. I currently have two units that I can't control with the PCB from esphome-lg-controller. I'm willing to assist if needed, including providing logs and being a guinea pig for testing.

Looking forward to seeing the updates based on the new document.

tolkachev, 2025-07-20:
Hey everybody! It seems i've managed to get working my LG ODU + 4 IDU + PI-485 + AtomLite&RS485 + HomeAssistant
if you wanna test it right away, pls check PR comment https://github.com/jourdant/esphome-lgap/pull/15
otherwise let's wait for @jourdant decision/merge/comments.
Have a nice day!

mystardious, 2026-03-11:
This was resolved by PR #15 — the PI-485 requires `tx_byte_0: 0x80` as the first byte in the TX sequence. This is now the default in the current codebase.

We've also added a dedicated "PI-485 Expansion Board" section to the README with DIP switch settings (1,4,5 ON), IDU addressing steps, and LED troubleshooting.

Thanks to everyone who contributed debugging info in this thread — especially @tolkachev for the fix and @TerryFrench, @greyburn, and @MaikaiLife for the extensive testing.