jourdant/esphome-lgap #12: Help wanted communicating with MULTI V IV outdoor unit

espcurious, 2025-01-07:
Hey anyone who had success with using project!

I'm dipping into this world with the target to read out and control my LG MultiV.IV airconditioning system.
I've rammed my head into a wall and need a bit of help how to progress.

I hope to get suggestions for steps to take to find out if some DIP switches need to be adjusted to enable communication on the ODU, and/or debug why no communication is flowing.

**What I have tried already**
Read all I could from this project, the issues in it, and all of the links I have spotted.

I think my setup resembles the one of @jourdant, but I have very little luck communicating with the ODU and finding documentation from LG on my unit.

**My setup**
My outdoor unit is
[image]
[image]

With the cover off, it looks like this:
[image]

I have the m5stack atom-lite with the RS485-base 
[image]
[image]

It is wired up like shown in the May 19th 2024 comment in 
https://github.com/jourdant/esphome-lgap/issues/1 

My ODU has the same connectors:
[image]

My yaml is pasted below, and I have tried all combinations of 
- swapped A and B wires 
- swapped rx and tx pins
- esp32-board-types
- baud-rates (2400, 4800, 9600, 19200, 57600, 115200)

<details>
<summary>esphome-lgap.yaml (expand)</summary>
```
esphome:
  name: esphome-lgap
  friendly_name: esphome-lgap

esp32:
#  board: m5stack-atom
  board: m5stack-core-esp32
  framework: 
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "WHCS6CnXpJbJ5ukiUBqyWtcOKhu6jIcpS8tDgjFRI0Y="

ota:
  - platform: esphome
    password: "936fa2e41698aba5805bcc889625cce1"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "esphome-lgap Fallback Hotspot"
    password: "WXGGA9MFPPs8"

captive_portal:

external_components:
  - source:
      type: git
      url: https://github.com/jourdant/esphome-lgap
      ref: main
    components: [ "lgap"]
    refresh: 0sec

uart:
  tx_pin: GPIO22
  #tx_pin: GPIO19
  rx_pin: GPIO19
  #rx_pin: GPIO22
  baud_rate: 4800
  data_bits: 8
  parity: NONE
  stop_bits: 1
  id: lgap_uart1

lgap:
  - id: lgap1
    uart_id: lgap_uart1

climate:
  - platform: lgap
    id: lgap_zone_1
    name: 'Zone 1 - Living Room'
    lgap_id: lgap1
    zone: 0
  - platform: lgap
    id: lgap_zone_2
    name: 'Zone 2 - Master Bedroom'
    lgap_id: lgap1
    zone: 1
  - platform: lgap
    id: lgap_zone_3
    name: 'Zone 3 - Kitchen'
    lgap_id: lgap1
    zone: 2
  - platform: lgap
    id: lgap_zone_4
    name: 'Zone 4 - Guest Bedroom'
    lgap_id: lgap1
    zone: 3
  - platform: lgap
    id: lgap_zone_5
    name: 'Zone 5 - KC'
    lgap_id: lgap1
    zone: 4
  - platform: lgap
    id: lgap_zone_6
    name: 'Zone 6 - CK'
    lgap_id: lgap1
    zone: 5
  - platform: lgap
    id: lgap_zone_7
    name: 'Zone 7 - Hallway'
    lgap_id: lgap1
    zone: 6
```
</details>
Any suggestions for things to try are most welcome!

espcurious, 2025-01-07:
Adding logs from esphome deployment while connected A->A & B->B with Tx set to GPIO19 and Rx to GPIO22 at baud-rate 4800 (but all attempts look the same, except for a warning when baud-rate is not set to 4800.

<details>
<summary>Output from code deployment</summary>

```
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/esphome-lgap.yaml...
INFO Updating https://github.com/jourdant/esphome-lgap@main
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-lgap (board: m5stack-core-esp32; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
Compiling .pioenvs/esphome-lgap/src/main.cpp.o
Linking .pioenvs/esphome-lgap/firmware.elf
RAM:   [=         ]  12.4% (used 40704 bytes from 327680 bytes)
Flash: [=====     ]  50.9% (used 933689 bytes from 1835008 bytes)
Building .pioenvs/esphome-lgap/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/esphome-lgap/firmware.bin"], [".pioenvs/esphome-lgap/firmware.elf"])
Wrote 0xf55c0 bytes to file /data/build/esphome-lgap/.pioenvs/esphome-lgap/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/esphome-lgap/firmware.bin"], [".pioenvs/esphome-lgap/firmware.elf"])
========================= [SUCCESS] Took 9.75 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.184.76 port 3232...
INFO Connected to 192.168.184.76
INFO Uploading /data/build/esphome-lgap/.pioenvs/esphome-lgap/firmware.bin (939456 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 5.67 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.184.76 using esphome API
INFO Successfully connected to esphome-lgap @ 192.168.184.76 in 11.225s
INFO Successful handshake with esphome-lgap @ 192.168.184.76 in 0.083s
[18:58:43][I][app:100]: ESPHome version 2024.12.2 compiled on Jan  7 2025, 18:58:17
[18:58:43][C][wifi:600]: WiFi:
[18:58:43][C][wifi:428]:   Local MAC: 90:15:06:FA:8E:54
[18:58:43][C][wifi:433]:   SSID: 'Bsus-iot'[redacted]
[18:58:43][C][wifi:436]:   IP Address: 192.168.184.76
[18:58:43][C][wifi:440]:   BSSID: 7E:45:58:37:2E:97[redacted]
[18:58:43][C][wifi:441]:   Hostname: 'esphome-lgap'
[18:58:43][C][wifi:443]:   Signal strength: -78 dB ▂▄▆█
[18:58:43][C][wifi:447]:   Channel: 6
[18:58:43][C][wifi:448]:   Subnet: 255.255.255.0
[18:58:43][C][wifi:449]:   Gateway: 192.168.184.1
[18:58:43][C][wifi:450]:   DNS1: 192.168.184.1
[18:58:43][C][wifi:451]:   DNS2: 1.1.1.1
[18:58:43][C][logger:185]: Logger:
[18:58:43][C][logger:186]:   Level: DEBUG
[18:58:43][C][logger:188]:   Log Baud Rate: 115200
[18:58:43][C][logger:189]:   Hardware UART: UART0
[18:58:43][C][uart.arduino_esp32:151]: UART Bus 1:
[18:58:43][C][uart.arduino_esp32:152]:   TX Pin: GPIO19
[18:58:43][C][uart.arduino_esp32:153]:   RX Pin: GPIO22
[18:58:43][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[18:58:43][C][uart.arduino_esp32:157]:   Baud Rate: 4800 baud
[18:58:43][C][uart.arduino_esp32:158]:   Data Bits: 8
[... 76 lines cut ...]
[18:58:48][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[18:58:48][D][lgap.climate:221]: Generating READ request message for zone 3...
[18:58:49][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[18:58:49][D][lgap.climate:221]: Generating READ request message for zone 4...
[18:58:49][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[18:58:49][D][lgap.climate:221]: Generating READ request message for zone 5...
[18:58:50][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[18:58:50][D][lgap.climate:221]: Generating READ request message for zone 6...
[18:58:50][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[18:58:50][D][lgap.climate:221]: Generating READ request message for zone 0...
[18:58:51][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[18:58:51][D][lgap.climate:221]: Generating READ request message for zone 1...
[18:58:52][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[18:58:52][D][lgap.climate:221]: Generating READ request message for zone 2...
[18:58:52][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[18:58:52][D][lgap.climate:221]: Generating READ request message for zone 3...
[18:58:53][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[18:58:53][D][lgap.climate:221]: Generating READ request message for zone 4...
[18:58:53][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[18:58:53][D][lgap.climate:221]: Generating READ request message for zone 5...
[18:58:54][E][lgap:127]: Last receive time exceeded. Clearing buffer...
[18:58:54][D][lgap.climate:221]: Generating READ request message for zone 6...
[...]
```
</details>

Anoxym, 2025-01-07:
For my system (UUD3 U30 & UM36F), I use RS485 module to control my LG system.

LG-RC-MBS-1
[image]

It allows to control the LG unit using register addresses (that I control with an esp32 that have RS485 interface)

Here is the address table that exposes what you can control and get for sensors : 

https://cdn.chipkin.com/assets/uploads/2018/feb/16-01-14-46_intesisbox_lg-rc-mbs-1_user_manual_en.pdf

This whole system works and I can do theses things : 

- set power on/off
- set mode (Auto, Cooling, Heating, Fan and dry)
- set Fan speed (Auto, 0, 1, 2, 3)
- set temperature target (16-30C)

- get errors
- get compressor status (on, to on, to off, off)
- get target temperature 
- get temperature from control panel
- get compressor running time
- get AC unit running time

espcurious, 2025-01-07:
Hi @Anoxym,
Is this box wired to the CEN A and CEN B terminals on the outdoor unit, or how is connected?

Anoxym, 2025-01-07:
It is wired on SIGNAL / GND / +12V

[image]

espcurious, 2025-01-07:
Okay, this is on the indoor unit bus.. 

My understanding was that the ESP with the m5-stack RS485-base would be able to connect directly to the CEN A+B connectors directly, so I'll still be aiming for that, but happy that another solution exists, should this not pan out.

Anoxym, 2025-01-07:
Have you tried to use SODU A & SODU B ?

espcurious, 2025-01-08:
No, haven't tried those. 
My understanding is that the CEN-terminals are for a central controller - like a BMS or a home assistant - and the SODU is for connecting multiple outdoor units in a master/slave constellation.

I found this 'install tips' document on scribd:
[image]
Not sure how to make the ODU set the signal high and low to measure the voltage on the terminals and what the values should be..